UnexpectedMaker/ReflowMaster #1: BOM and Schematic PDF

DirtyEngineer, 2018-08-11:
Great project. Thanks for sharing it!

Might I suggest listing a BOM of the purchased parts?

 Affiliate links are always welcome as well. 

Also I support your decision to add a "Buy Me A Coffee" link. So many times I've found and used an open source project and have no way of expressing my thanks to the person that put all the hard work into it.  I really wish GitHub had this integrated. 

I've attached a PDF of the current schematic (Rev F) as I find this super handy when checking out a project.

[ReflowMaster_Rev_F.pdf](https://github.com/UnexpectedMaker/ReflowMaster/files/2280601/ReflowMaster_Rev_F.pdf)

UnexpectedMaker, 2018-08-11:
Hey Jon,

That’s quite funny… I had a schematic PDF and a BOM txt file sitting in my local repo uncommitted - had totally forgotten to push it… I am glad I did forget though, the BOM was slightly out of date and still included the SWD Debug header that is no longer on the board.

Thanks for the reminder to push it :-)

I’m not a fan of using a BOM that is linked to a supplier as everyone gets their parts from different places, and I often will source a set of parts for a board from 3-4 places based on costs and availability of parts, so I’m just including a TXT based BOM that people can load into their preferred suppliers sites.

I agree that GITHUB should offer a built in “donate” option for free repos, but it wasn’t too much effort to get this setup - no takers yet though ;)

Keep fighting the good fight!

closed 2018-08-11 by UnexpectedMaker

DirtyEngineer, 2018-08-12:
Any chance you could include a little more detail as to specifically what screen is needed to work with this?

You would have had takers for your "donate" link if you hadn't threaten to refund donations during your live stream :)

UnexpectedMaker, 2018-08-12:
I've added links for AliExpress and eBay in the readme, and also added a bit about needing to install a bootloader on the SAMD21G before putting it on the board.

The people that were going to donate on my stream already support me by buying my products on tindie.. the donate button is for people that are not buying my products, but are instead using my open source files to make their own ;)