UnexpectedMaker/ReflowMaster #2: hermite tangents not supplied

tablatronix, 2018-08-15:
```C++
  baseCurve.setPoints(CurrentGraph().reflowGraphX, CurrentGraph().reflowGraphY, CurrentGraph().len);
  baseCurve.setDegree( Hermite );
```

I am surprised your environment doesn't crash because of this, esp32 does

You specify Hermite, but provide no tangents to _m
`void setPoints( float x[], float y[], float m[], int numPoints );`

in spline, this causes exceptions on esp32, loadprohibited , since _m is unset

```
     // calc()
     // _m[i], _m[i+1]
      return hermite( ((x-_x[i]) / (_x[i+1]-_x[i])), _y[i], _y[i+1], _m[i], _m[i+1], _x[i], _x[i+1] );
```

Oddly , The spline lib doesn't seem to catch this or ensure that _m is set for hermite sets..

No idea what memory this accesses on samd, maybe its safer, shrug.

tablatronix, 2018-08-15:
my workaround is to just add tangents array to the reflowGraph class
and  fill it with 0s

```C++
float tangents[10];
...
for ( int i = 0; i < minLength; i++ )
tangents[i]     = 0;
```

UnexpectedMaker, 2018-08-15:
Hmm, interesting that it doesn't crash on the SAMD... I'll update the codebase anyway to include a defualt set of tangents. Thanks!

closed 2018-08-15 by UnexpectedMaker