UnexpectedMaker/ReflowMaster #6: Bom don’t contain certain part names only what package

NoSkillsToPayTheBillz, 2019-05-28:
I can’t seen to figure out what voltage regulators you are using? The diode is also just the package type. Also the crystal is the same just the package. Kinda confused about the 12.5 pf 0805s also since they aren’t able to be bought. I’ve asked and searched a few suppliers and have been told such a part isn’t made.

UnexpectedMaker, 2019-05-28:
I just updated the repo with missing component names and changed 12.5pF caps for the XTAL to 12pF.
Cheers :-)

NoSkillsToPayTheBillz, 2019-05-28:
Thanks that makes it some much easier only thing is what crystal is it that your using

UnexpectedMaker, 2019-05-28:
The Crystal I am using is discontinued, so you'l need to just source a 32.768 crystal with the size specs from the BOM and use that with matching caps. There are stacks to choose from.,

NoSkillsToPayTheBillz, 2019-05-28:
I ordered a NX3215SA so that should work match’s your specs. Thanks for your time I know your super busy with the Pico crowd source.

closed 2019-05-28 by NoSkillsToPayTheBillz