UnexpectedMaker/ReflowMaster #7: abort immediate

tablatronix, 2019-07-06:
abort does not turn off heater immediatly

add
`SetRelayFrequency(0);`
do not wait for return and loop

```C++
void AbortReflow()
{
  if ( state == 1 || state == 2 || state == 16 ) // if we are in warmup or reflow states
  {
    SetRelayFrequency(0); // relay off
```

UnexpectedMaker, 2019-07-07:
This has been improved in v1.04 that I am going to upload today, but the abort button should never be able to be pressed outside of those states, so not sure why you need SetRelayFrequency(0); outside the if statement.

It is inside the if statement in v1.04 :-)

tablatronix, 2019-07-07:
It was a safety thing, in case of a bug or loss of state , abort aborts always

UnexpectedMaker, 2019-07-07:
Right, but again, AbortReflow() is only ever called from a button press, and that press can only happen in those states...

```
    else if ( state == 1 || state == 2 || state == 16 )
    {
      AbortReflow();
    }
```

tablatronix, 2019-07-07:
I am always thinking about future abstraction less code coverage if abort gets called from a big red button or tc failure

closed 2020-12-21 by UnexpectedMaker