UnexpectedMaker/ReflowMaster #14: Temperature controller questions

DurandA, 2020-12-19:
Hi @UnexpectedMaker and thanks for your open-source projects and YouTube videos.

I am making a pretty similar project for a [hot plate](https://aliexpress.com/item/32888343657.html) where I replace the original controller board with a custom one. Since there is just space for a segment display and some buttons, reflow profile will be configured from a smartphone using BLE. The [hardware](https://easyeda.com/DurandA/reflow) is mostly final and I am now working on the [firmware](https://github.com/DurandA/reflow946) using ESP-IDF.

I stumbled upon your project and found that you use the same thermocouple chip (MAX31855) than me and I am wondering how did you settle on the current temperature controller. I started working on a PID controller but saw that that you are only using a differential input:
https://github.com/UnexpectedMaker/ReflowMaster/blob/29f8a3fb8e1a09a3da2df962912d4bd50d375017/Code/Reflow_Master_v2/Reflow_Master_v2.ino#L997-L998

I did not try to tune the PID controller on the plate yet but was wondering how did you settle on a differential-only controller? Does the temperature closely match the chosen profile? I also saw some recommendations to rather use bang-bang (on-off) controllers for heaters due to the large inertia.

UnexpectedMaker, 2020-12-21:
I don't think PID is a good fit for reflow, so I implemented my own solution. It's not really differential control, and more a combo of a few techniques, that allows users to tune in a more understandable way.
But this really isn't a RM issue, so maybe my discord server is a better place to talk about it 👍 
https://discord.com/invite/xAHpApP

closed 2020-12-21 by UnexpectedMaker

DurandA, 2020-12-21:
Whoops, I meant proportional controller.

Sorry for opening an issue and thanks for the Discord link!