UnexpectedMaker/ReflowMaster #17: Support esp32?

liwajaya, 2021-10-31:
Thanks for sharing great Project. I need to know this code support for esp32. I need to buikd dIY  but SAM IC dificult to find in my town. Thanks

UnexpectedMaker, 2021-10-31:
No it doesn't, sorry. It won't compile as is.
You could modify it and remove the parts that don't work on the ESP32 and integrate new code that does.
If you do, remember to make it public and open source as as well.

anthonysleck, 2021-11-16:
> Thanks for sharing great Project. I need to know this code support for esp32. I need to buikd dIY but SAM IC dificult to find in my town. Thanks

I agree it would be cool to have. I have started looking at it for an ESP32, but I am seeing how great of a task this will be. Will let you know if I get some more time to put something together.

anthonysleck, 2021-11-16:
> No it doesn't, sorry. It won't compile as is. You could modify it and remove the parts that don't work on the ESP32 and integrate new code that does. If you do, remember to make it public and open source as as well.

BTW Great project. Nice work.

closed 2021-12-19 by UnexpectedMaker