UnexpectedMaker/ReflowMaster #20: Shortage of ATSAMD21G-A?

Bazmundi, 2021-12-19:
Currently looks like ATSAMD21G-A are hard to source.   How many of the SAM D21 family might be an optional drop-in replacement please?  Based on, I guess, why you opted for that chip.

https://au.mouser.com/c/?q=ATSAMD21G seem to have various SAM D21G variants.

UnexpectedMaker, 2021-12-19:
Reflow Master only supports the ATSAMD21G18A - in the QFP package.

Anything else will require you to modify the PCB design (if you go for a QFN package of the 21G18A) or require bootloader and code changes to work on a version with less Flash/SRAM.

UnexpectedMaker, 2021-12-19:
It's in stock at mouser... though not many.
https://au.mouser.com/ProductDetail/Microchip-Technology-Atmel/ATSAMD21G18A-AUT?qs=sGAEpiMZZMtXHE36kCvv36ajhsTJC%252B%2FBgmvnKz0leZ0%3D

You only need 1 right?

Bazmundi, 2021-12-19:
Ta, I'll get me 10 but missed that in the search - too literal.

Bazmundi, 2021-12-19:
DOH, can't get the K-type  MAX31855K, have to wait (apparently)

andygrillo, 2022-03-27:
ATSAMD21G18A in QFP seems to be hard to find with the main distributers. Given that I only need 1 for this project, and I don't have a programmer to load arduino, I thought to buy this board from aliexpress:

https://www.aliexpress.com/item/1005003445206496.html?spm=a2g0o.productlist.0.0.7a916ee2h9RW5J&algo_pvid=37a7e5d5-9eef-42c4-806d-6dfa0d4050d2&algo_exp_id=37a7e5d5-9eef-42c4-806d-6dfa0d4050d2-5&pdp_ext_f=%7B%22sku_id%22%3A%2212000025820256134%22%7D&pdp_pi=-1%3B254.07%3B-1%3B95.53%40salePrice%3BMXN%3Bsearch-mainSearch

It appears from the fuzzy photo to have the ATSAMD21G18A in QFP package, and I can erasily upload the bootloader using the board. Once done I can hot gun the chip off the board and use it for the reflow oven PCB.

What do you think?

UnexpectedMaker, 2022-03-27:
Sure, using a doner chip from another board is fine, so long as you have the tools and skills to remove it and re-solder without killing it.

andygrillo, 2022-03-27:
great, ill do that then. I would assume a hot air gun should do the trick.

closed 2023-03-29 by UnexpectedMaker