UnexpectedMaker/ReflowMaster #21: Updated Board Files?

dmadison, 2022-04-20:
I ordered a board from you last year (thank you, by the way!) and it's labeled "2020-2". It looks like the EAGLE files in the repo are labeled "2018-3" and are missing the programing header.

Is there any chance you could update the board files in the repo to the latest version?

UnexpectedMaker, 2022-04-20:
Hi David, the files I'm keeping full open source are the original design files.
Is there a reason you need the newer files?

dmadison, 2022-04-20:
Thanks for the quick response, Seon.

I'm working on putting the controller, relay, and everything else into its own fully contained enclosure. As part of that I was planning to build an add-on button board so that the buttons face the same direction as the screen. I was hoping to use the design files as a position and wiring reference.

UnexpectedMaker, 2022-04-20:
The buttons are the same from a wiring point of view. Boards are identical in functionality and still run the same code. Spacing on buttons etc are also still the same. The board grew in the other direction just a little, so you should be able to reference the online version here for what you want to do.

dmadison, 2022-04-20:
Oh, I didn't realize it had grown in size as well. I'm planning on building a mounting bracket for the LCD / controller sandwich, so that might have an effect.

I don't want to make more work for you, but would you consider sharing a vector model of the board (DXF/SVG) with just the edge outline and drill points? That would be all I need for a CAD reference.

UnexpectedMaker, 2022-04-20:
This is the best I can do atm:
[ReflowMaster_Outline.pdf](https://github.com/UnexpectedMaker/ReflowMaster/files/8518353/ReflowMaster_Outline.pdf)

If you cant get what you need from this, I can look at something else on the weekend.

dmadison, 2022-04-20:
I can definitely make that work. Thanks!

closed 2022-04-20 by dmadison

dmadison, 2022-04-21:
In case it helps anyone, here's a cleaned up SVG version of the above file. I removed the vias and made the holes accurate to size (more or less):

[image]