UnexpectedMaker/ReflowMaster #22: Which USB socket?

pastaclub, 2023-03-23:
Some BOM items are hard to identify, especially the Micro USB socket. What is the exact component used?

I tried two different components but they do not exactly fit the footprint. I noticed the two holes in the PCB, presumably for little bolts in the component (which would actually be very useful for manual pick and place).

Also, which exact MCU is used? The BOM just says ATSAMD21G-A, but there are dozens of variations of that chip in a 48pin package. Does it matter or can I pick any?

ns2gupta, 2023-03-29:
I think the design changed a lot since it was posted. I see that the new version uses ESP32.

@UnexpectedMaker, can you please share the updated schematic and board layout?

UnexpectedMaker, 2023-03-29:
> I think the design changed a lot since it was posted. I see that the new version uses ESP32.
> 
> @UnexpectedMaker, can you please share the updated schematic and board layout?

Reflow Master hasn't changed design.

The ESP32 version is Reflow Master Pro, a different product and not open source.

UnexpectedMaker, 2023-03-29:
> Some BOM items are hard to identify, especially the Micro USB socket. What is the exact component used?
>
> I tried two different components but they do not exactly fit the footprint. I noticed the two [...]

SAMD21G18A is the version you need. The specific USB connector is no longer available, nor can I find it listed, so you'll need to source a matching one, or alter the PCB to use a different connector and get the boards re-made.

pastaclub, 2023-04-01:
Thank you for the support!

closed 2023-04-01 by pastaclub