UnexpectedMaker/ReflowMaster #24: Reflow Master Pro ESP32 Code ¿?

jlabordaversa, 2024-01-13:
It is Reflow Master Pro ESP32 code available open source? I wanted to add some improvements to my oven.

Mike-Menci, 2024-01-13:
Yes Go ahead!

UnexpectedMaker, 2024-01-13:
No, reflow Master Pro is not open source.

UnexpectedMaker, 2024-01-13:
> It is Reflow Master Pro ESP32 code available open source? I wanted to add some improvements to my oven.

What functionality did you want added?

jaimelaborda, 2024-01-14:
Thank you very much for your reply.

I wanted to add PWM for fan outputs and also ideally I wanted to be able to turn the graphical interface 90 degrees so I can use it vertically as it would better fit my toaster oven. I am also thinking about changing the display to an smaller one.
But of course, if the project is not open source I cannot. I hoped that if the first version was open source this second one with the ESP32 would also be open source.

> > It is Reflow Master Pro ESP32 code available open source? I wanted to add some improvements to my oven.
> 
> You want to add functionality to your Reflow Master Pro? Or you want to add functionality from the Reflow Master Pro to your own non Reflow Master Pro setup?
> 
> Can you be more specific? What functionality did you want added?

UnexpectedMaker, 2024-01-14:
@jaimelaborda AUX output is already PWM - You have the ability to set the FREQ and the % is the duty cycle.

If you can source a 2.2" ILI9341 screen that's 320x240, you can just switch the screens out, but that's not much smaller. Going smaller also drops the resolution and that's not compatible with the firmware.

Screen rotation or alternate resolution is not possible as the code is written fixed for 320x240 in the specific orientation the RMP ships in.

closed 2025-01-08 by jlabordaversa